sciguy14/Remote-Wake-Sleep-On-LAN-Server #9: Flush empty buffer

smark91, 2015-05-27:
in index.php:

change:
```php
ob_end_flush();
```
with:
```php
if (ob_get_level()>1) {ob_end_flush();}
```

avoid flushing empty buffer (on x-debug it make an error).

sciguy14, 2015-09-15:
@smark91 When I did this, it resulted in no dynamic flushing of the current state - that's undesirable. What is the error that you're seeing? I don't get any errors.

smark91, 2015-09-21:
This is the x-debug error I have.

[image]

sciguy14, 2015-10-29:
@smark91 Are actually experiencing functionality problems because of this? What you're described is a notice, not an error. Attempting to flush an empty buffer shouldn't have any effect on the programs functionality, and making the change you recommended makes flushing fail entirely. I'm closing this for now

closed 2015-10-29 by sciguy14