sciguy14/Remote-Wake-Sleep-On-LAN-Server #10: apache2.service fails to start on OSMC

HeedfulCrayon, 2015-10-28:
I have followed all the steps in the readme file, but I am still encountering an issue with starting the apache2 service.  
[image]

sciguy14, 2015-10-29:
Another user found that he needed to install libapache2-mod-php5

What happens if you do that using this command:
`sudo apt-get install libapache2-mod-php5`
Then restart apache?

You can also check the apache2 logs to get a better idea of what's going on:
`cat /var/log/apache2/error.log`

sciguy14, 2015-10-29:
@felixrr might also be able to provide some insight once you share your logs.

changty, 2015-11-15:
I'm actually having the same problem on newest Raspbian.

cat /var/log/apache2/error.log
```
[Sun Nov 15 10:52:45.270379 2015] [mpm_event:notice] [pid 17392:tid 3069514256] AH00489: Apache/2.4.10 (Raspbian) configured -- resuming normal operations
[Sun Nov 15 10:52:45.274114 2015] [core:notice] [pid 17392:tid 3069514256] AH00094: Command line: '/usr/sbin/apache2'
[Sun Nov 15 10:53:51.804673 2015] [mpm_event:notice] [pid 17392:tid 3069514256] AH00491: caught SIGTERM, shutting down
[Sun Nov 15 10:53:54.349631 2015] [mpm_prefork:notice] [pid 18411] AH00163: Apache/2.4.10 (Raspbian) configured -- resuming normal operations
[Sun Nov 15 10:53:54.354445 2015] [core:notice] [pid 18411] AH00094: Command line: '/usr/sbin/apache2'
[Sun Nov 15 10:54:11.043107 2015] [mpm_prefork:notice] [pid 18411] AH00169: caught SIGTERM, shutting down
[Sun Nov 15 10:54:15.071871 2015] [mpm_prefork:notice] [pid 18514] AH00163: Apache/2.4.10 (Raspbian) configured -- resuming normal operations
[Sun Nov 15 10:54:15.073600 2015] [core:notice] [pid 18514] AH00094: Command line: '/usr/sbin/apache2'
[Sun Nov 15 11:02:20.429517 2015] [mpm_prefork:notice] [pid 18514] AH00169: caught SIGTERM, shutting down
```
I'm also missing the directory conf.d mentioned later on README.md: 
/etc/apache2/conf.d/security

Please let me know, if I can provide any additional log to help catching this bug.

changty, 2015-11-15:
I changed the /etc/apache2/mods-enabled/ssl.conf line "SSLMutex..." to just "Mutex..." and created a directory /var/run/apache2/ssl_mutex. After this I got the server running but the SSL is not (yet) working.

commit 80cc99f7 (2016-01-25): At v2.0. Updated for Apache 2.4.x and other fixes.

* Closes #11 by adding a note to the README about setting proper permissions for pings from the Pi.
* Closes #10 by finally getting everything up to snuff with the newest Apache .conf file formats.
*Closes #12 and other HTTPS/HTTP issues that were highlighted on my blog,
by make the the README instructions much more clear, fixing config problems, and making SSL mod loading only execute when SSL is actually enabled.
* Cleans everything up in general. Both HTTP and HTTPS configurations are tested and confirmed working on a Fresh Raspberry Pi Mo [... cut]
closed 2016-01-25 by sciguy14