sciguy14/Remote-Wake-Sleep-On-LAN-Server #11: No Ping Response

Zuendapp, 2015-12-19:
Hi,

i have used Remote-Wake-Sleep-On-LAN-Server over two years but since a few weeks the ping says "Still Down" and PC is asleep. The wake succeeds but the response is wrong. A ping from my Pi to PC works. Do you have any idea?

Thank you very much!

Best regards

sciguy14, 2015-12-31:
Hmm. What OS is the machine that you are waking up? Did you recently upgrade or make any other network changes in the last few weeks?

The Wake-Features depends on the MAC address, but the ping feature depends on the IP. So, if the IP changed, but the NIC remained the same (thus, same MAC) you might see this issue.

Zuendapp, 2016-01-07:
Hi,
Windows 10. I'have moved my folder from /var/www to a virtual machine with debian OS and it works, but I have the same problem with a second RaspberryPi. I have also changed in config.php the IP Adress to my routers IP which is always on, but the state is also ".... is presently asleep". A normal ping to all of my IP addresses works :(.

Best regards

commit 80cc99f7 (2016-01-25): At v2.0. Updated for Apache 2.4.x and other fixes.

* Closes #11 by adding a note to the README about setting proper permissions for pings from the Pi.
* Closes #10 by finally getting everything up to snuff with the newest Apache .conf file formats.
*Closes #12 and other HTTPS/HTTP issues that were highlighted on my blog,
by make the the README instructions much more clear, fixing config problems, and making SSL mod loading only execute when SSL is actually enabled.
* Cleans everything up in general. Both HTTP and HTTPS configurations are tested and confirmed working on a Fresh Raspberry Pi Mo [... cut]
closed 2016-01-25 by sciguy14