sciguy14/Remote-Wake-Sleep-On-LAN-Server #13: PHP Error

Volpey, 2016-01-28:
hey, i tried to setup your WOL on my computer in the past few days, sometimes i get it to work, but right now the website is emtpy, i guess there is an error somewhere in the config but i can't find it.

> PHP Warning:  Cannot modify header information - headers already sent by (output started at /var/www/html/index.php:23) in /var/www/html/index.php on line 35

thanks in advance!

sciguy14, 2016-02-01:
Two bug fixes that I pushed yesterday may fix your header issue. Go grab the latest commit - specifically, update your .htaccess file and 000-default.conf files to the latest ones from my repo. You can see the changes in the commit history. Don't forget to restart apache after updating: `sudo service apache2 restart`. Let me know if that helps.

sciguy14, 2016-02-16:
Did my update fix your issue?

Klapsa2503, 2016-04-28:
Your fix does not work. Actually this helps: `?computer=0`.
I wish I could help but i am not a php developer. If you will have any further questions feel free to ask me :)

sciguy14, 2016-09-10:
The latest also might have fixed this: https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server/pull/25

closed 2016-09-10 by sciguy14