sciguy14/Remote-Wake-Sleep-On-LAN-Server #15: White,Empty web page

EinderJam, 2016-01-31:
Sorry,but i'm still encountering errors with this project.but first,thank you for your update.
So i reinstalled raspbian on my raspberry pi 2 model B,to have a fresh install.
I followed your new,up-to-date instructions,and everything went fine,i put my DNS in the "common name ",and leaved the password line empty.
And after,when i tried to access the page,Firefox warned me about the certificate,i confirmed the exception,and i accessed the webpage.
Webpage was empty,not like in my previous issue,but really empty,just white.
Thank you,EinderJam

P.S:when i edit  the config file,do i have to remove the " ,when i enter the hash,for exemple ?

sciguy14, 2016-02-01:
Hi @EinderJam -

First off, I've added a screenshot of what a correct config file should look like to the wiki. Go take a look at it and make sure that you're quotes are correct: https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server/wiki/Installation#finishing-the-configuration

To diagnose the empty pages, it will helpful to see your error logs. Go look at the error logs in var/log/apache2. Do you see anything of interest in error.log and ssl_error.log? If so, please share your errors here.

EinderJam, 2016-02-02:
thank you for your answer,and for your screenshot ! It's very helpful !
the error was mine,i forgot the " before and after the hash,and now the site works,but i can't access it with the DNS,but i'll look for what i missed......
Thank you for your great job again !!!!

sciguy14, 2016-02-16:
Glad to hear you worked it out. Good luck with the dynamic DNS config.

closed 2016-02-16 by sciguy14