sciguy14/Remote-Wake-Sleep-On-LAN-Server #16: Problem with ping?

carlosg85, 2016-02-23:
Hi, 
all is running perfectly, but i cannot do it from the front-end web. 

It always say me "Querying Computer State. Please Wait..." and then that my copmputer  is presently awake. False! It's power-off, but never see the 'wake' button. always 'sleep' button.

I can wakeonlan from the console with the mac without problems.

What is the problem with the ping command? thanks!!

carlosg85, 2016-02-24:
I've modified the index.php to do it works
In the $pinginfo comparation with empty ($pinginfo == "" or != "", i've to put this text "1 packets transmitted, 0 packets received, 100% packet loss".

Now, it works perfectly!

ewyll, 2016-03-04:
Thanks @carlosg85, it works with that fix :).

closed 2021-02-21 by sciguy14