sciguy14/Remote-Wake-Sleep-On-LAN-Server #18: No space left on device

tonysmith86, 2016-04-22:
$ git clone https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server.git
fatal: could not create work tree dir 'Remote-Wake-Sleep-On-LAN-Server'.: No space left on device

Sorry i am pretty new to this and was keen on setting up a remote WOL server however when i try to clone the repository it returns i have no space left on the device. This is a 32GB micro SD card that i just installed Jessie on so find it hard to believe i have used 32GB already. i had the same on an 8GB SD card so went to 32GB and got the same.

Any help would be appreciated, and apologies in advance if this is a stupid question.

alexruffell, 2016-04-22:
This may be a long shot but it is my opportunity to help for once.... when you first setup the RPI did you expand the partition to use the entire uSD card? This is one of the first things you need to do once you flash the card with the image otherwise you will actually end up not using the entire card... run `sudo raspi-config` and select "1 Expand Filesystem".

tonysmith86, 2016-04-22:
That was exactly it. I have successfully configured it however now get a blank window when i hit the URL after accepting the certificate. Any tips on what to check?

Sorry this is all a little new to me.

closed 2016-04-22 by tonysmith86