sciguy14/Remote-Wake-Sleep-On-LAN-Server #19: Can sleep but cant wake

tonysmith86, 2016-04-22:
Hi i have managed to get this installed and configured however having issues waking my machine. If i use a separate WOL utility, i.e. iOS iNet WOL on my LAN i can get the machine to wake.

I am able to hit the web server externally but when i send the wake command the machine never powers up but i cant make it sleep.

I have tried this on the LAN bypassing port forwarding of my router etc and the utility still does not wake the machine.

Any help would be appreciated.

tonysmith86, 2016-04-25:
If someone could direct me to a log file or something to see what is stopping the wake i would appreciate it.

alexruffell, 2016-04-25:
I was having your same issue when I first set it up but I am having trouble recalling what it was... so this is a shot in the dark... try to verify that the RPI is actually able to wake up your computer by running the same command it uses in its code (exec ('wakeonlan ' . $COMPUTER_MAC[$selectedComputer]);)

If running wakeonlan xx:xx:xx:xx:xx:xx works then it may be due to the configuration of the app. However, I do not believe this is the right spot to get support/help on getting this up and running as Github issues are meant to report problems with the code. May I recommend you try a forum or the comments section that already has 282 support related comments?

http://www.jeremyblum.com/2013/07/14/rpi-wol-server/#comments

Good luck! Wish I could help you more!

tonysmith86, 2016-04-25:
Thanks for the reply, this actually gave me the hint i needed to rectify
the issue, it was the MAC format... i used - instead of : schoolboy error.

Works a treat tho!!

closed 2016-05-01 by sciguy14