sciguy14/Remote-Wake-Sleep-On-LAN-Server #20: SSL external issue

ClaytonAstrom, 2016-04-23:
Not sure if this is a bug or an issue with my firewall.  Port 443 is open, port 5000 redirects to port 80 local.

Checking the /var/log/apache2/error.log, I get the following:

> PHP Notice: Undefined index: HTTPS in /var/www/html/index.php on line 24

Which is:

> if ($USE_HTTPS && !$_SERVER['HTTPS']){
> header("Location: https://$_SERVER[HTTP_HOST]$_SERVER[REQUEST_URI]");
> exit;
> }

Followed the instructions (https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server/wiki/Installation) exactly, and I can access the site on a local computer, so it may not be a bug at all.  However, I'm not sure what could have gone wrong with port forwarding, since it's fairly straight forward

sciguy14, 2016-05-01:
That means the HTTPS variable is not defined in the $_SERVER global variable. Sounds like a PHP configuration issue. What is the URL you're using to access the site? Is it: http://yoursite.com:5000? What happens why you try to access the HTTPS site directly? https://yoursite.com

ClaytonAstrom, 2016-05-02:
@sciguy14 thanks for the response!

Using chrome, the url I'm using is just the one from No-IP (not specifying http or https, letting it resolve on it's own).  I have the domain name setup to go to :5000, and then my router to redirect 5000 to 80.  Whether I put in http or https though, the client response is the same:  "The site can't provide a secure connection":
<details> 
  <summary>Screen shot</summary>
   <p>[image]</p> 
</details>

I think the ports are getting screwed with somehow.  Here's the full error log:
`[Mon May 02 12:37:19.477972 2016] [:error] [pid 683] [client <my phone's global ip>:4095] PHP Notice:  Undefined index: HTTPS in /var/www/html/index.php on line 24`

I'm getting different ports for some reason, whether I use the FQDN or even the IP address directly with the associated port.  I get the same response locally, however, so maybe that's nothing.  From a local computer (which, despite the error, allows me to connect):
`[Mon May 02 12:50:41.005029 2016] [:error] [pid 687] [client 192.168.1.129:50389] PHP Notice:  Undefined index: HTTPS in /var/www/html/index.php on line 24`

Hope this information helps, let me know what else I can provide.  Thanks!

closed 2021-02-21 by sciguy14