sciguy14/Remote-Wake-Sleep-On-LAN-Server #23: Multiple NICs

Pappardello, 2016-07-18:
Hi,

I have successfully installed all of the needed software and it seems to be "working" but there is one big issue: the pc doesn't turn on. I tried connecting to the RBPI via vnc and issuing a wakeonlan command and it is working but the web interface does not wake up my PC.

This is my setup: RBPI connected via WiFi to the router and via ethernet cable to the PC. 
I noticed that the wakeonlan software works only if i specify the destination IP address to 192.168.137.1 (ip of the pc's cable NIC while ON, rbpi's cable static ip: 192.168.137.2) could it be that your software is not choosing the right interface to send the WOL packet on even though i did set up the correct destination address in the config file? If so, then an option to select the WOL output interface (or ip) would be great.

Thanks :)

wigald, 2017-02-11:
Did you find a solution here ? I got the same problem now. My RPI ist connected via openvpn to get a static wan ip for mailserver purposes. Since then this is thing isn't working anymore :( I need to provide a broadcast address (equal to commandline: wakeonlan -i 192.168.1.255 00:00:00:00), this is working like a charm with wakeonlan via script. Didn't find something in these files so far.

wigald, 2017-02-11:
alright, I found it :D

try it like this:

`nano +181 index.php`

change this `exec ('wakeonlan ' . $COMPUTER_MAC[$selectedComputer]);`

to this:

`exec ('wakeonlan -i 192.168.178.255 ' . $COMPUTER_MAC[$selectedComputer]);`

Change your subnet broadcast address to yours

This works to me :)

closed 2021-02-21 by sciguy14