sciguy14/Remote-Wake-Sleep-On-LAN-Server #24: OS X El Capitan: need Bonjour Sleep Proxy?

beeb, 2016-08-25:
First of all thanks for this great utility! I set it up very easily and it works like a charm for my windows computer.

Now I have a MacBook Pro running El Capitan on the same network that I tried adding to the config file (with static local IP and mac address double checked) but I can't get it to wake by magic packet.

I made sure the mac has it's "Wake for Wifi network access" option enabled and set up a fixed local IP in the wifi adapter settings. When I ssh into my pi from the windows computer then run `wakeonlan xx:xx.....` with the MAC address of the MacBook Pro, nothing happens.

From what I have read, this might be due to the fact that Wake On Demand for Apple products require an Airport Express router or Apple TV because they run a "Bonjour Sleep Proxy"..?
Is there a way to make this work with the Raspberry Pi or should I give up?

Thanks

sciguy14, 2016-09-10:
As far as I know (Apple or not), Wake-on-LAN is only possible over a wired interface - I don't think you can send a Magic Packet to a Wi-Fi connected machine.

roolebo, 2020-04-11:
Not sure how I landed on the issue, but JFYI I have seen a MAGICWAKE event in pmset assertions on a Wi-Fi interface of my laptop:
```$ networksetup -listallhardwareports
..snip...
Hardware Port: Wi-Fi
Device: en0
...cut...
$ pmset -g assertions
..snip...
Kernel Assertions: 0x104=USB,MAGICWAKE
   id=504  level=255 0x100=MAGICWAKE mod=06.04.2020, 21:22 description=en0 owner=en0
...snip...
```

closed 2021-02-21 by sciguy14