sciguy14/Remote-Wake-Sleep-On-LAN-Server #27: Win10 Anniversary Update Breaks WOL/SOL Functionality

Eyeheartpie, 2016-11-07:
I've made no changes to my system other than updating to Win10 Anniversary, and now the WOL/SOL functionality doesn't work. I can get to the webpage, but attempting WOL just tries to wake, pings 15 times to check status, and then stops with failure. I even turned on my computer and then tried to shut it down with the sleep on lan, and it says that the computer is asleep on the status page. It was working 100% fine until this most recent major Win10 update. Do you have any ideas on what may have changed to cause this to break?

Thanks!

aloureiro, 2016-11-07:
Hi! It's working well for me with Win10 Pro (fully updated).
Did you turn off  "fast boot" and turn on the firewall rule for ping reply?

Eyeheartpie, 2016-11-07:
I had it turned off, and it was still off. I didn't change the firewall rule for ping reply and it used to work. Maybe now I just need to manually add that. I'll give it a shot and see if it works.

Eyeheartpie, 2016-11-08:
Turns out it was a twofold issue: Win10 Anniversary update replaced the manufacturer ethernet driver with an MS driver, and firewall rule for ping reply was disabled. I uninstalled that driver and reinstalled the one direct from my mobo's site, and then enabled ping reply, and now it all works again.

closed 2016-11-08 by Eyeheartpie