sciguy14/Remote-Wake-Sleep-On-LAN-Server #34: Changing the port from port 80

tuckaluck, 2017-10-20:
Sorry if this is in the wrong spot.

Is there any way to change the port from port 80 after installation?

I'm currently running pihole on the RPI as well and would like to keep that on port 80?

sciguy14, 2017-10-20:
If you follow the https instructions, that will put it on port 443.

Jeremy Blum
Blum Idea Labs, LLC.
http://www.jeremyblum.com

Sent from my phone, please excuse the brevity.

tuckaluck, 2017-10-22:
Thanks, this should have been obvious. Can I start the installation steps again from the start of the SSL instructions without any issue?

sciguy14, 2017-10-22:
You should be able to. Give it a shot

Jeremy Blum
Blum Idea Labs, LLC.
http://www.jeremyblum.com

Sent from my phone, please excuse the brevity.

cartersm, 2019-01-20:
For future travelers: If you want the apache server to listen on a port other than 80, follow [these instructions](https://www.cyberciti.biz/faq/linux-apache2-change-default-port-ipbinding/).

closed 2021-02-17 by sciguy14