sciguy14/Remote-Wake-Sleep-On-LAN-Server #36: Error installing git and apache2 (Failed to resolve)

keem85, 2017-12-04:
Hi guys

I get error when I try to install the packages. I get a message that it fails to resolve to raspbian.trivini.no.. I have tried several mirrors and update/upgrade command, but still I get this error.. Is this project abandoned?

Kim

sciguy14, 2017-12-04:
That sounds like an issue with the package repos that your pi is setup to
use. I don't control those packages. Is your pi running stock raspbian?

Jeremy Blum
Blum Idea Labs, LLC.
http://www.jeremyblum.com

Sent from my phone, please excuse the brevity.

keem85, 2017-12-04:
Hi schiguy14. You were right! I had to edit the repository differently than I did. I got everhting working now except for one thing.. I only have the sleep button. Do you have any idea why I see only the sleep button?

keem85, 2017-12-04:
It says that "The computer is presently awake", even though it is really not.. I tried it with two different PC's, both of them are returning with a "Sleep" button. It is really strange.

keem85, 2017-12-04:
Got it all working now :) It was just a matter of waiting. Sleep does also work, except that the program does not start with windows. Put it in shell:startup and also activated it via task manager, but still it doesn't start.. But I see others have reported this. But that's a problem with the Sleep on Lan application, not with the webservice :)

closed 2017-12-04 by keem85