sciguy14/Remote-Wake-Sleep-On-LAN-Server #41: [Suggestion] Add instructions for implementing Certbot

FriedMax, 2019-12-17:
Self-signed certificates are a fantastic way to add security to this project, and I liked getting to sign my own certificate. However, there is a tool called Certbot that automatically installs a signed certificate from a free CA called Let's Encrypt. Perhaps there could be a mention of it in the tutorial with hints on how to get going with it?

sciguy14, 2021-01-04:
I agree. I will work on implementing certbot.

sciguy14, 2021-02-17:
@FriedMax Checkout the v3 beta branch, where I've built a comprehensive setup script that basically does everything for you, including installation of the signed cert from Let's Encrypt.
Branch: https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server/tree/v3_beta
Instructions: https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server/wiki/Installation-(v3-beta)

sciguy14, 2021-02-21:
Closing issue because the v3_beta branch adds certbot implementation. @FriedMax please feel free to re-open this ticket if you try out the v3_beta branch and certbot doesn't work for you.

closed 2021-02-21 by sciguy14