sciguy14/Remote-Wake-Sleep-On-LAN-Server #43: Created an Docker Image, Please add a Link to the Image/Repo

ex0nuss, 2020-12-26:
Hi sciguy14,

I am using your software for more than a year and I am a huge fan of it.
In the last weeks, I wrote an proper Image with environment variables, documentation and more. 
In the first line of the documentation I gave you credits for the software :)

It would be nice, if you can add a Link in your README / Documentation:
- GitHub: https://github.com/ex0nuss/Remote-Wake-Sleep-On-LAN-Docker
- Dockerhub: https://hub.docker.com/r/ex0nuss/remote-wake-sleep-on-lan-docker

Regards,
ex0nuss

closed 2021-01-04 by sciguy14

sciguy14, 2021-01-04:
Done, thanks for making this Docker image! You're timing is great, because I was actually working with Docker for an unrelated project, and was thinking about making a Docker image for this. You'll find a link on the readme, and also on my blog: https://www.jeremyblum.com/2013/07/14/rpi-wol-server/

ex0nuss, 2021-01-04:
Thx so much for adding it to the Readme and blog! 
The image gets automatically updated on every sunday ^^ 
Just a quick note: The docker image does not run on a Raspberry Pi since it's build for x86.

sciguy14, 2021-01-04:
Good point, I updated to note to point out that its an x86 Docker image.

ex0nuss, 2021-01-05:
Perfect! 
If there are any Issues from users regarding docker, forward them to my repo ^^

commit a92c933c (2021-01-04): Add link to Docker Image
Fixes #43