sciguy14/Remote-Wake-Sleep-On-LAN-Server #46: Website says "Desktop is currently asleep" even though it isn't.

DwertlePlayz, 2021-04-15:
The website shows "desktop is currently asleep" even though it is not. I double-checked and the config file is correct, the port forwarding is done, and the pi is connected to the router through ethernet (so is the pc). Did I miss anything?
[image]

DwertlePlayz, 2021-04-15:
I'm not too sure if it helps, but the pi is connected to the router directly through ethernet while the computer is connected through an ethernet switch that is connected to the router.

mihcox, 2022-06-24:
Having the same issue myself, it does work to start the computer but it cannot tell if its online.

Using in docker, with RPi pihole as dhcp.

sciguy14, 2023-02-05:
This means that ping requests are not being replied. You will need to make sure the computer you are trying to control is configured to allow ICMP ping requests. Many have this _disabled_ by default. See here: https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server/wiki/Installation-(v3)#setting-up-the-remote-machines

closed 2023-02-05 by sciguy14