sciguy14/Remote-Wake-Sleep-On-LAN-Server #47: snap "core" is not available on stable for this architecture (armel)

prapador, 2021-07-24:
Hello, first, thank you very much for your work.

During the installation of RWSOLS V3, I have received the following error:

`snap "core" is not available on stable for this architecture (armel) but exists on other architectures (amd64, arm64, armhf, i386, ppc64el, s390x).`

I am using a Raspberry Pi Zero

Thanks a lot!

dontwatchthisspace, 2021-08-03:
Not sure if it helps, but I've received this same issue on a Raspberry Pi Zero as well, and a Raspberry Pi Model B Rev 2.

sciguy14, 2023-02-06:
Sorry, but the snap store doesn't have support for the older instruction sets. You may need to use a newer Pi that is is on the ARMv8-A instruction set (see here: https://en.wikipedia.org/wiki/Raspberry_Pi#Specifications).

Alternatively, you can modify the setup script to use apt versions of the snap apps I am using, but I used the snap apps because i had trouble getting the apt versions to behave properly.

closed 2023-02-13 by sciguy14