sciguy14/Remote-Wake-Sleep-On-LAN-Server #52: Can this be used for local networks only?

SBCV-apegram, 2022-07-31:
I don't need to be able to remotely access this outside of the building (I'll be using a VPN for that). I do like the idea of a nice interface for waking up a media machine that a user inadvertently shut down.

sciguy14, 2022-07-31:
Yes, you can use it like that. Simply skip the port forwarding and dynamic DNS setup if you don't need those.

SysCrasher, 2022-08-23:
That's possible, like @sciguy14 said (Thanks btw for your work, it's much more comfortable than do the magic through ssh; especially on mobile) 
If you don't need a password, insert hash `e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855` in config.php.
Then the password is empty and you only need to click the button.

closed 2023-02-05 by sciguy14