sciguy14/Remote-Wake-Sleep-On-LAN-Server #60: No longer works with no password

stuckj, 2023-04-30:
I've forked ex0nuss's repository (https://github.com/ex0nuss/Remote-Wake-Sleep-On-LAN-Docker) as it's no longer maintained. While trying out the fork I kept running into problems with the password being denied even though I didn't set a password (so APPROVED_HASH = NULL).

After a little debugging, I think this is due to changes in `www/html/index.php` in this repo back on Feb 5 to prevent information leakage (pre-password entry). `approved`, `wake_up`, `go_to_sleep`, and `check_current_status` are all set to false initially and are only ever changed in the block that runs when `is_null($APPROVED_HASH)` is false (meaning you have a password set). This means that with no password set those will all remain false and you'll end up getting to the else block on line 281 showing an invalid password error.

I verified that with a password set it works fine.

stuckj, 2023-04-30:
Also, I have a forked repo based on ex0nuss's that will be updated weekly from my teamcity server if you'd like to reference it in addition to ex0nuss's. This one: https://github.com/stuckj/Remote-Wake-Sleep-On-LAN-Docker. It's fully working except for the no password issue mentioned above.

sciguy14, 2023-05-02:
Thanks for sharing his info here for others, but I'm closing this issue because this is working as intended, as far as I'm concerned. A password is necessary for the secure implementation of this service, so I don't want to support a version that doesn't require one.

closed 2023-05-02 by sciguy14

stuckj, 2023-05-02:
Faire enough. I hadn't realized that was a feature from his repository only. Thanks.