sciguy14/Remote-Wake-Sleep-On-LAN-Server #69: pi 5

eskile, 2024-02-22:
the setup script does not work on pi 5 running raspbian bookworm.
Package libapache2-mod-php7.4 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'php7.4' has no installation candidate
E: Unable to locate package php7.4-curl
E: Couldn't find any package by glob 'php7.4-curl'
E: Couldn't find any package by regex 'php7.4-curl'
E: Package 'libapache2-mod-php7.4' has no installation candidate
Error installing prerequisite software.
Script Exiting due to failure in 'Install Prequisite Software' setup step.

sciguy14, 2024-02-25:
Fixed by #70 - PHP version detection is now automatic during installation, so it should work on Raspbian Bookworm.

closed 2024-02-25 by sciguy14

eskile, 2024-03-07:
Hello! 
Have you tested the setup script on a pi 5?
I have specifically  a pi 5 8gb model. I've tested the install of the newest releasy on both normal 64bit bookworm and the "lite" version suggested in the installation guide. If it is to any help i have my wifi region set to sweden(se) and same for locale settings.
Here's the install feedback:
Running: Install Prequisite Software
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
/bin/sh: 1: Syntax error: Unterminated quoted string
Error installing prerequisite software.
Script Exiting due to failure in 'Install Prequisite Software' setup step.

my user is the default pi.