sciguy14/Remote-Wake-Sleep-On-LAN-Server #71: Pi 5 setup script

eskile, 2024-03-09:
Hello! 
I saw that you updated the code with version detection, however I am still having some problems.
Have you tested the setup script on a pi 5?
I have specifically  a pi 5 8gb model. I've tested the install of the newest release on both normal 64bit bookworm and the "lite" version suggested in the installation guide. If it is to any help i have my wifi region set to sweden(se) and same for locale settings.
Here's the install feedback:

Running: Install Prequisite Software
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
/bin/sh: 1: Syntax error: Unterminated quoted string
Error installing prerequisite software.
Script Exiting due to failure in 'Install Prequisite Software' setup step.

My user is the default pi with default password.

viggeSandberg, 2024-03-11:
I had the same issue!
I ended up disabling lines 140-142 in "setup.py" and changed line 145 to the following: "subprocess.run(['sudo', 'apt-get', '-y', 'install', 'wakeonlan', 'git', 'apache2', 'php8.2, 'php8.2-curl', 'libapache2-mod-php8.2', 'snapd'], check=True)", is now installing as it should :)

eskile, 2024-03-11:
Hello! I do not know how to edit python files. could you add the updated script to your comment?

viggeSandberg, 2024-03-11:
No problem! 
I am also having issues with permissions, I get a 403 forbidden when I try to access the webpage. I'll try some thing this evening and I'll come back ;)

setup.py:

#! /usr/bin/python3

# Remote Wake/Sleep-On-LAN Server [SETUP SCRIPT]
# https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server
# (c) 2024 Blum Idea Labs, LLC.
# Author: Jeremy E. Blum (https://www.jeremyblum.com)
# License: GPL v3 (http://www.gnu.org/licenses/gpl.html)

# Import Libraries
import subprocess
import pathlib
import os
import signal
import sys
import urllib.request
import urllib.error
import re
import getpass
import socket
import time
import ssl
import fileinput

# Exit Handler
def signal_handler(sig, frame):
    print('\n\nABORTING SCRIPT.\n')
    sys.exit(1)
signal.signal(signal.SIGINT, signal_handler)

# Global Variables
script_dir = pathlib.Path(__file__).parent.absolute()
public_ipv4 = "127.0.0.1"
urls = []
encryption_mode="none" # Options: "none","self","certbot","skip"
ddns_temp_config=pathlib.Path('/tmp/ddclient.conf')
ddns_real_config=pathlib.Path('/etc/ddclient.conf')
rwsols_config_sample = script_dir.joinpath('www/html/config_sample.php')
rwsols_config_user = script_dir.joinpath('www/html/config.php')

# Colors
black = lambda text: '\033[0;30m' + text + '\033[0m'
red = lambda text: '\033[0;31m' + text + '\033[0m'
green = lambda text: '\033[0;32m' + text + '\033[0m'
yellow = lambda text: '\033[0;33m' + text + '\033[0m'
blue = lambda text: '\033[0;34m' + text + '\033[0m'
magenta = lambda text: '\033[0;35m' + text + '\033[0m'
cyan = lambda text: '\033[0;36m' + text + '\033[0m'
white = lambda text: '\033[0;37m' + text + '\033[0m'

# Main Logic
def main():
    # Intro
    print("")
    print(cyan(">>>                                                       <<<"))
    print(cyan(">>> Remote-Wake-Sleep-On-LAN-Server (RWSOLS) Setup Script <<<"))
    print(cyan(">>>                                                       <<<\n"))

    # Warn the user that internet scripts are dangerous
    print(yellow("This script will install the required packages and configure permissions as necessary for RWSOLS to work properly. It assumes you are on the bookworm release."))
    print(yellow("As with all scripts that you find on the internet, I strongly suggest you read its contents before blindly executing it."))
    print(yellow("This script assumes your user is a passwordless sudo user, which is the default for the 'pi' user on Raspbian. It executes commands with sudo.\n"))
    print(yellow("If you want to abort this script to inspect its contents, press Ctrl+C now."))
    input(yellow("If you have read and understood what this script is doing, Press Enter to continue with the automatic setup."))
    print("")

[... 732 lines cut ...]
        else:
            https_server = True
            if txt.startswith("hello?"):
                https_rwsols = True
                print(magenta("Your RWSOLS is accessible on port 443, but it is using a self-signed SSL certificate!"))
            else:
                print(yellow("Something was found at " + urls[0] + " port 443, but it was not RWSOLS."))
    except ssl.SSLError as e:
            print(yellow("The following SSL error was encountered while trying to verify the server:"))
            print(yellow(str(e)))
    else:
        https_server = True
        https_signed = True
        if txt.startswith("hello?"):
            https_rwsols = True
            print(magenta("Your RWSOLS is accessible on port 443 and is using a properly signed SSL certificate!"))
        else:
            print(yellow("Something was found at " + urls[0] + " port 443, but it was not RWSOLS."))

    
    return http_server, http_rwsols, https_server, https_rwsols, https_signed 

# Execute
if __name__ == "__main__":
    main()

eskile, 2024-03-23:
Have you gotten the permissions to work? once i visit the site it gives me the error :Forbidden
You don't have permission to access this resource.

viggeSandberg, 2024-03-25:
I ended up removing the symlink Remote-Wake-Sleep-On-LAN-Server/www/html -> /var/www/ and moved the data from the first folder to the /var/www/. See v2 [documentation](https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server/wiki/Installation-(v2---Deprecated)#setting-up-the-apache-2-server-and-securing-it) as a startingpoint.

andretdias, 2024-05-07:
Hi there, running into the same issue.

Could you provide a more noob friendly step by step to fix the :Forbidden error on the website?

lukee12zakk, 2024-05-29:
can I have help pls not working on pi 4 with rasbian 64bit lite.
my setup file looks like this:

#! /usr/bin/python3

# Remote Wake/Sleep-On-LAN Server [SETUP SCRIPT]
# https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server
# (c) 2024 Blum Idea Labs, LLC.
# Author: Jeremy E. Blum (https://www.jeremyblum.com)
# License: GPL v3 (http://www.gnu.org/licenses/gpl.html)

# Import Libraries
import subprocess
import pathlib
import os
import signal
import sys
import urllib.request
import urllib.error
import re
import getpass
import socket
import time
import ssl
import fileinput

# Exit Handler
def signal_handler(sig, frame):
    print('\n\nABORTING SCRIPT.\n')
    sys.exit(1)
signal.signal(signal.SIGINT, signal_handler)

# Global Variables
script_dir = pathlib.Path(__file__).parent.absolute()
public_ipv4 = "127.0.0.1"
urls = []
encryption_mode="none" # Options: "none","self","certbot","skip"
ddns_temp_config=pathlib.Path('/tmp/ddclient.conf')
ddns_real_config=pathlib.Path('/etc/ddclient.conf')
rwsols_config_sample = script_dir.joinpath('www/html/config_sample.php')
rwsols_config_user = script_dir.joinpath('www/html/config.php')

# Colors
black = lambda text: '\033[0;30m' + text + '\033[0m'
red = lambda text: '\033[0;31m' + text + '\033[0m'
green = lambda text: '\033[0;32m' + text + '\033[0m'
yellow = lambda text: '\033[0;33m' + text + '\033[0m'
blue = lambda text: '\033[0;34m' + text + '\033[0m'
magenta = lambda text: '\033[0;35m' + text + '\033[0m'
cyan = lambda text: '\033[0;36m' + text + '\033[0m'
white = lambda text: '\033[0;37m' + text + '\033[0m'

# Main Logic
def main():
    # Intro
    print("")
    print(cyan(">>>                                                       <<<"))
    print(cyan(">>> Remote-Wake-Sleep-On-LAN-Server (RWSOLS) Setup Script <<<"))
    print(cyan(">>>                                                       <<<\n"))

    # Warn the user that internet scripts are dangerous
    print(yellow("This script will install the required packages and configure permissions as necessary for RWSOLS to work properly. It assumes you are on the bookworm release."))
    print(yellow("As with all scripts that you find on the internet, I strongly suggest you read its contents before blindly executing it."))
    print(yellow("This script assumes your user is a passwordless sudo user, which is the default for the 'pi' user on Raspbian. It executes commands with sudo.\n"))
    print(yellow("If you want to abort this script to inspect its contents, press Ctrl+C now."))
    input(yellow("If you have read and understood what this script is doing, Press Enter to continue with the automatic setup."))
    print("")

    # Add note about skipping completed steps
    print("Note: One-time steps that have already been executed will be automatically skipped.")
[... 729 lines cut ...]
                https_rwsols = True
                print(magenta("Your RWSOLS is accessible on port 443, but it is using a self-signed SSL certificate!"))
            else:
                print(yellow("Something was found at " + urls[0] + " port 443, but it was not RWSOLS."))
    except ssl.SSLError as e:
            print(yellow("The following SSL error was encountered while trying to verify the server:"))
            print(yellow(str(e)))
    else:
        https_server = True
        https_signed = True
        if txt.startswith("hello?"):
            https_rwsols = True
            print(magenta("Your RWSOLS is accessible on port 443 and is using a properly signed SSL certificate!"))
        else:
            print(yellow("Something was found at " + urls[0] + " port 443, but it was not RWSOLS."))

    
    return http_server, http_rwsols, https_server, https_rwsols, https_signed 

# Execute
if __name__ == "__main__":
    main()

I get this error when installing using help form a comment on this thread form user viggeSandberg.
[image]

lukee12zakk, 2024-05-29:
just tryed and getting this:

[image]

lukee12zakk, 2024-05-30:
managed to get it working by removing the php code in the setup.py file but when try to acess the website that apache is on i get this

[image]

itzmattu, 2024-06-09:
I went through the script and fixed what was going wrong with this error.

[Fixing setup.py improper handling of process output](https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server/pull/73)

Be advised that this still resulted in a 403 Forbidden error when trying to reach the web server. I simply deleted the symlink created at /var/www and copied the contents of $HOME/RWSOL/www to /var/www instead, and then it worked great.

PR #73 (2024-12-22): Fixing setup.py improper handling of process output
Properly handling the output from the prerequisites package search processes when fed into package installation command lines.

fixes #71 
fixes #72
closed 2024-12-22 by sciguy14