sciguy14/Remote-Wake-Sleep-On-LAN-Server #72: "Error installing prerequisite software" (key is stored in legacy trusted.gpg keyring)

MarcoPo10, 2024-05-17:
I've been trying to get this running on a Pi Zero 2W, getting this same error during the setup.py step every time. I've flashed every version of Raspberry Pi OS Lite (bookworm) from March 2024 going back to December 2023 with the same results. Since the default username and password have been removed from the OS since early 2022 I set up the image with username "pi" and password "raspberry" for the install. I don't think that's causing the issue here. I also checked the URL included in the error as well as opening the /etc/apt/trusted.gpg file. The URL was a dead-end and the gpg file was scrambled. 

Error:
_Running: Install Prequisite Software
Hit:1 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Hit:2 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done_                             
**W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
/bin/sh: 1: Syntax error: Unterminated quoted string
Error installing prerequisite software.
Script Exiting due to failure in 'Install Prequisite Software' setup step.**

[image]

[image]

lukee12zakk, 2024-05-23:
i have the same problem exept a bit different
[image]

JTricky, 2024-06-03:
I'm getting the same error as @lukee12zakk except mine is erroring out after reading the package lists of Hit : 4

itzmattu, 2024-06-09:
Same error as the OP. I'm on a Raspberry Pi 2B, fresh install but setup a user/pass (instructions mention script relies on no password for sudo, but I assume it would just popup a sudo pw prompt.

itzmattu, 2024-06-09:
I went through the script and fixed what was going wrong with this error.

https://github.com/sciguy14/Remote-Wake-Sleep-On-LAN-Server/pull/73

Be advised that this still resulted in a 403 Forbidden error when trying to reach the web server. I simply deleted the symlink created at /var/www and copied the contents of $HOME/RWSOL/www to /var/www instead, and then it worked great.

PR #73 (2024-12-22): Fixing setup.py improper handling of process output
Properly handling the output from the prerequisites package search processes when fed into package installation command lines.

fixes #71 
fixes #72
closed 2024-12-22 by sciguy14